puuu/ESPiLight #2: Working with NanoESP Board?

Terr4, 2016-11-07:
Hi,

I try to get the library working with my NanoESP Board which is basically an Arduino Nano + ESP8266 (http://www.mikrocontroller-elektronik.de/wifi-board-nanoesp-bzw-pretzel-board/)

I am not an Arduino expert and so far I programmed it with the Arduino IDE with board setting "Arduino Nano".
When I load the Example "Basic ESPiLight transmit example" and try to send it to the board I get the error:

`WARNING: library ESPiLight claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [avr] architecture(s).
In file included from /Users/Terra/Documents/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/alecto_ws1700.c:26:0:
/Users/Terra/Documents/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/../../core/dso.h:25:2: error: unknown type name 'ssize_t'
  ssize_t size;
  ^
exit status 1
Error compiling for board Arduino Nano.`

When I switch the board in the IDE to "Generic ESP8266 Board" I get the error:

`Sketch uses 301,281 bytes (69%) of program storage space. Maximum is 434,160 bytes.
Global variables use 46,956 bytes (57%) of dynamic memory, leaving 34,964 bytes for local variables. Maximum is 81,920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed`

What am I doing wrong?...

br

puuu, 2016-11-08:
Hi,
thank you for trying ESPiLight.

Unfortunately, ESPiLight can not run on AVR architecture, its is made to run native on the ESP8266 SoC. The NanoESP Board utilizes an Atmel ATmega 328 (AVR)  and only communicate with the ESP8266 via RX/TX. You may can flash the ESP8266 of the NanoESP Board by utilizing the 8-pin header and an USB-RX/TX module directly.

I would suggest to buy a ESP8266 board, like Adafruit Feather HUZZAH, NodeMCU or wemos d1 mini. IMHO they are much more powerful and cheaper, than the NanoESP Board.

> When I load the Example "Basic ESPiLight transmit example" and try to send it to the board I get the error:
```
WARNING: library ESPiLight claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [avr] architecture(s). In file included from /Users/Terra/Documents/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/alecto_ws1700.c:26:0: /Users/Terra/Documents/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/../../core/dso.h:25:2: error: unknown type name 'ssize_t' ssize_t size; ^ exit status 1 Error compiling for board Arduino Nano.
```

ESPiLight can not run on AVR architecture.

> When I switch the board in the IDE to "Generic ESP8266 Board" I get the error:
```
Sketch uses 301,281 bytes (69%) of program storage space. Maximum is 434,160 bytes.
Global variables use 46,956 bytes (57%) of dynamic memory, leaving 34,964 bytes for local variables. Maximum is 81,920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
```
> What am I doing wrong?...

The compilation runs fine, but the Arduino IDE was not able to flash it. That is because you did not connected the ESP8266 for flashing.

puuu

Terr4, 2016-11-09:
Thank you very much for your detailed explanation!! :)
I will try it with an adafruit board.

br

Terr4, 2016-11-17:
Just wanted to say thank you for this library.
I am using it now on a NodeMCU and it works like a charm for my Brennenstuhl RCS 1000 N plugs!

puuu, 2016-11-29:
Thank you for using ESPiLight.
Nice to hear someone is using it.

Happily close this issue.

closed 2016-11-29 by puuu