puuu/ESPiLight #3: Handling protocol "aliases"

robertklep, 2016-12-21:
From what I understand from the pilight sources, device protocols can be referenced through different names.

For instance, the `arctech_switch_old` protocol has an alias `kaku_switch_old`, as defined [here](https://github.com/pilight/pilight/blob/20742babc7700e40f6858d007de061f242617dbe/libs/pilight/protocols/433.92/arctech_switch_old.c#L218).

However, using the alias doesn't work with ESPiLight: it will only allow the actual name of the protocol to be used (in this case, `arctech_switch_old`). Using the alias generates an empty pulse train.

It took me a while to realize this while trying to get my old KaKu switch to work. I used the alias, as it's mentioned [here](https://wiki.pilight.org/doku.php/arctech_switch_old_v7_0) as protocol name.

puuu, 2016-12-21:
Thanks for reporting that aliases are not supported. Indeed, the function `protocol_device_add()` was replaced by a dummy function ([protocol.h](https://github.com/puuu/ESPiLight/blob/master/src/pilight/libs/pilight/protocols/protocol.h#L116)). This is by intention to reduce the memory usage of ESPiLight. Only the protocol id is supported, please use this.

robertklep, 2016-12-21:
Okay cool, thanks for the info!

closed 2016-12-21 by robertklep