puuu/ESPiLight #4: ESP-8266 (NODEMCU ESP-12E) Crash

ripper121, 2017-01-19:
With your Example code for Receive_raw and Receive my ESP is crashing after some seconds:

```
Exception (0):
epc1=0x4021098c epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc30 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdd0:  401070cc 3ffeced0 3ffeced0 402109a8  
3ffffde0:  ffffffff 00000020 00000001 00000000  
3ffffdf0:  00000000 4010142f 00000000 00000022  
3ffffe00:  3fffc200 40107094 3fffc258 4000050c  
3ffffe10:  4000437d 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 ffffffff 80000000  
3ffffe30:  20000000 3fff6a10 80000000 203fc0e0  
3ffffe40:  80000000 3fffc6fc 3fff3334 3fff6a14  
3ffffe50:  00000214 003fc0e0 60000600 00000030  
3ffffe60:  3fffc200 40107094 3fffc258 00000001  
3ffffe70:  40107130 00000002 00000000 4010711a  
3ffffe80:  ffffffff 00000020 08000000 070d0000  
3ffffe90:  00000000 00000000 0000001f 401055bd  
3ffffea0:  4000050c 40107094 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 00000014 ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  feefeffe 3fff3334 3fff3334 fffffffe  
3ffffee0:  00000210 3fff1828 3fff3334 3fffdab0  
3ffffef0:  00000000 3fffdad0 3fff34ac 00000030  
3fffff00:  00000000 400042db 3ffec771 60000600  
3fffff10:  40004b31 3fff6934 000002f4 003fc000  
3fffff20:  40105b22 3fff1850 3fff08c0 4010756c  
3fffff30:  40215739 3fff08c0 3fff1850 00ffb031  
3fffff40:  3fff6934 00001000 40215bd6 00000008  
3fffff50:  00000000 00000000 40215c83 3fff0974  
3fffff60:  3fff1850 004fbed9 3fff1850 4020fa24  
3fffff70:  40227e0d 3fff0974 3fff1850 00ffad8a  
3fffff80:  40227e52 3fffdab0 00000000 3fffdcb0  
3fffff90:  3fff1870 3fffdad0 3fff34ac 402117a7  
3fffffa0:  40000f49 00003976 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vc2f0c112
~ld
```

puuu, 2017-01-24:
I can not reproduce this issue. Is there no additional output, only the stack trace?
Did the examples `Receive` and `test_parse` show the same issue?

May be, you receive some data that cause an error of the pilight part during parsing. So please try to run without connected receiving module.

Unfortunately, I can not decode the stack trace, because this require binary files generated at build time. Please have a look to [EspExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder) and may post the results.

ripper121, 2017-01-24:
I use the NodeMCU 1.0 With ESP-12E.
I used your default Receive.ino
It looks you have a problem with the garbage collector:

```
pulses: 132 possible protocol: arctech_switch
FreeMem:22152 
pulses: 132 possible protocol: arctech_screen
FreeMem:22080 
pulses: 132 possible protocol: arctech_contact
FreeMem:22000 
pulses: 50 possible protocol: logilink_switch
FreeMem:21928 
pulses: 50 possible protocol: heitech
pulses: 50 possible protocol: ev1527
FreeMem:21912 
pulses: 50 possible protocol: eurodomest_switch
pulses: 50 possible protocol: elro_800_switch
pulses: 50 possible protocol: elro_800_contact
FreeMem:21760 
pulses: 50 possible protocol: daycom
FreeMem:21640 
pulses: 50 possible protocol: logilink_switch
Valid message: [logilink_switch] {"systemcode":21521,"unitcode":5,"state":"on"}
FreeMem:21712 
pulses: 50 possible protocol: heitech
pulses: 50 possible protocol: ev1527
Valid message: [ev1527] {"unitcode":769711,"state":"opened"}
FreeMem:21768 
pulses: 50 possible protocol: eurodomest_switch
pulses: 50 possible protocol: elro_800_switch
pulses: 50 possible protocol: elro_800_contact
Valid message: [elro_800_contact] {"systemcode":3,"unitcode":11,"state":"opened"}
FreeMem:21688 
pulses: 50 possible protocol: daycom
Valid message: [daycom] {"id":1,"systemcode":5137,"unit":5,"state":"on"}
FreeMem:21640 
pulses: 50 possible protocol: logilink_switch
FreeMem:21712 
pulses: 50 possible protocol: heitech
pulses: 50 possible protocol: ev1527
FreeMem:21768 
pulses: 50 possible protocol: eurodomest_switch
pulses: 50 possible protocol: elro_800_switch
pulses: 50 possible protocol: elro_800_contact
FreeMem:21688 
pulses: 50 possible protocol: daycom
FreeMem:21640 
pulses: 50 possible protocol: logilink_switch
FreeMem:21712 
pulses: 50 possible protocol: heitech
pulses: 50 possible protocol: ev1527
FreeMem:21768 
pulses: 50 possible protocol: eurodomest_switch
pulses: 50 possible protocol: elro_800_switch
pulses: 50 possible protocol: elro_800_contact
FreeMem:21688 
pulses: 50 possible protocol: daycom
FreeMem:21640 
pulses: 50 possible protocol: logilink_switch
FreeMem:21712 
pulses: 50 possible protocol: heitech
pulses: 50 possible protocol: ev1527
FreeMem:21768 
pulses: 50 possible protocol: eurodomest_switch
pulses: 50 possible protocol: elro_800_switch
pulses: 50 possible protocol: elro_800_contact
FreeMem:21688 
pulses: 50 possible protocol: daycom
FreeMem:21640 
[... 110 lines cut ...]
3ffffed0:  feefeffe 3fff3344 3fff3344 fffffffe  
3ffffee0:  00000210 3ffec464 3fff3344 3fffdab0  
3ffffef0:  00000000 3fffdad0 3fff34bc 00000030  
3fffff00:  00000000 400042db 3ffee867 60000600  
3fffff10:  40004b31 3fff6c14 000002f4 003fc000  
3fffff20:  40105b22 3fff1860 3fff08d0 4010756c  
3fffff30:  402155d9 3fff08d0 3fff1860 00b65724  
3fffff40:  3fff6c14 00001000 40215a76 00000008  
3fffff50:  00000000 00000000 40215b23 3fff0984  
3fffff60:  3fff1860 0038d38a 3fff1860 4020fa24  
3fffff70:  40227cad 3fff0984 3fff1860 00b649fa  
3fffff80:  40227cf2 3fffdab0 00000000 3fffdcb0  
3fffff90:  3fff1888 3fffdad0 3fff34bc 40211663  
3fffffa0:  40000f49 00018363 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vc2f0c112
~ld
```

ripper121, 2017-01-24:
Yes if i remove the Receiver from the Pin then there is no Problem.
If i connect it the after some seconds i get a reboot.

puuu, 2017-01-24:
Thank you for the detailed report! This looks like that a long and deep debugging session of the single pilight protocols is necessary... Please fell free to have a deeper look.

puuu, 2017-01-24:
You can try to remove some pilight protocols by removing their initialization from `src/pilight/libs/pilight/protocols/protocol_init.h`.

ripper121, 2017-01-24:
Im trying it allready. I left only arctech left but same problem

ripper121, 2017-02-08:
When I'm Using the git version the problem is solved.
2.3.0 has still some Interrupt issues, so it was not a problem of your lib.

closed 2017-02-08 by ripper121

puuu, 2017-02-09:
Did you mean 2.3.0 of the ESP8266 core for Arduino?
Are there any bug reports about the Interrupt issues?

I have ESPiLight running in [MQTT433gateway](https://github.com/puuu/MQTT433gateway) compiled with 2.3.0 of the ESP8266 core for Arduino. I did not observed any issue. But maybe it is a hardware issue, because I use a  Adafruit HUZZAH ESP8266.