puuu/ESPiLight #5: example 'transmit' using 2nd delay()

Jekyll555, 2017-02-07:
First of all Thanks! for all the cool code stuff. Got my rc swtich controlled in minutes after my RF sender postal arrived... that included internet search to find your lib!
My 2cts: 

  rf.send("elro_800_switch", "{\"systemcode\":17,\"unitcode\":1,\"on\":1}");
  delay(2000);
  rf.send("elro_800_switch", "{\"systemcode\":17,\"unitcode\":1,\"off\":1}");
  delay(2000); //<-- makes that blinking last~

ripper121, 2017-02-08:
What do you mean with "delay(2000); //<-- makes that blinking last~" please describe it a litte better

puuu, 2017-02-09:
Thank you for reporting this bug.

Fixed in e6a0dbb5f9d37ec08906623aeb8efd5c79faa534.

Nice to hear that other are using this library.

closed 2017-02-09 by puuu

ripper121, 2017-02-09:
What have you changed?

puuu, 2017-02-09:
@ripper121 please have a look to the commit e6a0dbb5f9d37ec08906623aeb8efd5c79faa534. I added the missing delay at the end of the loop in the transmit example.
So, it will be avaiable if I build a new release.