puuu/ESPiLight #7: clarus_switch receive working, send not

SciLor, 2017-04-26:
I have following RF-Signals:

```
RF signal arrived [clarus_switch][A3] (0) {"id":"A3","unit":12,"state":"off"}
RF signal arrived [clarus_switch][B3] (0) {"id":"B3","unit":12,"state":"off"}
RF signal arrived [clarus_switch][C7] (0) {"id":"C7","unit":12,"state":"off"}
RF signal arrived [clarus_switch][D11] (0) {"id":"D11","unit":12,"state":"off"}
RF signal arrived [clarus_switch][E3] (0) {"id":"E3","unit":13,"state":"off"}
```

when I try to send it via

```
  rf.send("clarus_switch", "{\"id\":\"A3\",\"unit\":12,\"state\":\"off\"}");
  rf.send("clarus_switch", "{\"id\":\"B3\",\"unit\":12,\"state\":\"off\"}");
  rf.send("clarus_switch", "{\"id\":\"C7\",\"unit\":12,\"state\":\"off\"}");
  rf.send("clarus_switch", "{\"id\":\"D11\",\"unit\":12,\"state\":\"off\"}");
  rf.send("clarus_switch", "{\"id\":\"E3\",\"unit\":13,\"state\":\"off\"}");

  rf.send("clarus_switch", "{\"id\":\"A3\",\"unit\":12,\"state\":\"on\"}");
  rf.send("clarus_switch", "{\"id\":\"B3\",\"unit\":12,\"state\":\"on\"}");
  rf.send("clarus_switch", "{\"id\":\"C7\",\"unit\":12,\"state\":\"on\"}");
  rf.send("clarus_switch", "{\"id\":\"D11\",\"unit\":12,\"state\":\"on\"}");
  rf.send("clarus_switch", "{\"id\":\"E3\",\"unit\":13,\"state\":\"on\"}");
```
I just get an error
```
piLightCreatePulseTrain: protocol: clarus_switch
LOG(3): clarus_switch: insufficient num
 create Code failed.
```

puuu, 2017-04-28:
Thanks for using ESPiLight!

Unfortunately, the JSON format for sending and receiving is different in pilight.

Please try something like this:
```c
rf.send("clarus_switch","{\"id\":\"A3\",\"unit\":12,\"on\":\"1\"}");
rf.send("clarus_switch","{\"id\":\"B3\",\"unit\":12,\"off\":\"1\"}");
```

SciLor, 2017-04-28:
Great, thank you, the output was a bit missleading.

closed 2017-04-28 by SciLor