puuu/ESPiLight #8: memory leak

gschintgen, 2017-06-04:
Hi,
I think there may be a memory leak in EsPiLight. (I'm not an expert developer, so it could well be some kind of user error...)
Here's a minimal code sample illustrating the problem I've encountered:

```
#include <ESPiLight.h>
#define TRANSMITTER_PIN D4
ESPiLight rf(TRANSMITTER_PIN);

#define runAtMostEvery(t) for (static uint16_t _lasttime;\
      (uint16_t)((uint16_t)millis() - _lasttime) >= (t);\
      _lasttime = (uint16_t)millis())

const char strOn[]  = "{\"systemcode\":11,\"unitcode\":1,\"on\":1}";

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
}

void loop() {
  runAtMostEvery(1000) {
    Serial.printf("loop heap size: %u\n", ESP.getFreeHeap());
    rf.send("elro_800_switch", strOn);
  }
}

```
And here's the corresponding output:
```
loop heap size: 27336
piLightCreatePulseTrain: protocol: elro_800_switch create Code succeded.
loop heap size: 27088
piLightCreatePulseTrain: protocol: elro_800_switch create Code succeded.
loop heap size: 26840
piLightCreatePulseTrain: protocol: elro_800_switch create Code succeded.
loop heap size: 26592
```
The free heap size is decreasing until the microcontroller crashes.

In my case it's a Wemos D1 Mini. I'm using Atom/PlatformIO.

puuu, 2017-06-05:
Thanks, good catch!

The minimal code sample was very helpful.

The bug is fixed in c7247d86abf80957a151a7a976e5b4dc74aa17c8 .

I will prepare a new release.

puuu, 2017-06-06:
[v0.11.1](https://github.com/puuu/ESPiLight/releases/tag/v0.11.1) is released, should be soon available in the arduino library manager.

closed 2017-06-06 by puuu