puuu/ESPiLight #10: Missing functions

kryptoss, 2017-06-25:
Hello,
I get this error on using the example "Receive" of the library installed from arduino library manager:

/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/quigg_gt1000.c.o: In function `createCode':
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/quigg_gt1000.c:325: undefined reference to `putchar'
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/quigg_gt1000.c:341: undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/quigg_gt9000.c.o: In function `createEncryptedData':
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/quigg_gt9000.c:247: undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/quigg_gt7000.c.o: In function `createCode':
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/quigg_gt7000.c:215: undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/quigg_gt7000.c.o: In function `createParity':
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/quigg_gt7000.c:197: undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/quigg_screen.c.o:/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/quigg_screen.c:232: more undefined references to `putchar' follow
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/core/json.c.o: In function `sb_finish':
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/core/json.c:396: undefined reference to `calloc'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/core/json.c.o: In function `mknode':
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/core/json.c:396: undefined reference to `calloc'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/alecto_ws1700.c.o: In function `checkValues':
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/alecto_ws1700.c:141: undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/alecto_ws1700.c.o:(.text.parseCode+0x80): undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/alecto_wsd17.c.o:(.text.parseCode+0x5c): undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/alecto_wx500.c.o:(.text.parseCode+0x70): undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/arctech_contact.c.o: In function `parseCode':
/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/arctech_contact.c:69: undefined reference to `putchar'
/tmp/arduino_build_82692/libraries/ESPiLight/pilight/libs/pilight/protocols/433.92/arctech_dimmer.c.o:/home/user/Arduino/libraries/ESPiLight/src/pilight/libs/pilight/protocols/433.92/arctech_dimmer.c:91: more undefined references to `putchar' follow
collect2: error: ld returned 1 exit status

It seems several functions are missing. 
Thank you
Regards

puuu, 2017-06-25:
What version of Arduino and ESP8266 Board module you are using and what doard did you have selected?

I can compile it without errors with:
Arduino: 1.8.3
ESP8266: 2.3.0
Boards: Generic ESP8266 Modul/Adafruit HUZZAH 8266/Wemos D1 R1 & mini

puuu, 2017-12-05:
No answer for long time. Close it. Please reopen if not solved.

closed 2017-12-05 by puuu