puuu/ESPiLight #13: Trying to send Arctech_Switch Signal

Dennifly, 2017-11-26:
Hi, im am building an Alexa All in ONE ESP8266 Controll Unit.

I have 3 Device's i read aout with Pilight on Raspian . But i can not send the code via my NodeMCU v3.

I load the example Scatch and this works but when i change to this 

rf.send("kaku_switch", "{\"systemcode\":46008320,\"unitcode\":1,\"on\"-t}");

it says always 

piLightCreatePulseTrain: invalid json ("sys....":46008320,unit .. .. . . a.s.o.

i did not find the Mistake . i have the code the unit and state on and off ?

Can someone help me . Thanks

Dennifly, 2017-11-26:
I solved the Problem. for Arctech_Switch use this

  rf.send("arctech_switch", "{\"id\":46008320,\"unit\":1,\"on\":1}");
  delay(2000);
  rf.send("arctech_switch", "{\"id\":46008320,\"unit\":1,\"off\":1}");
  delay(2000);

:-D

puuu, 2017-11-28:
Thanks for using ESPiLight. Is this issue solved and can be closed?

Dennifly, 2017-12-02:
Thanks for your mail i solved the Problem . Thank you !

Von: puuu
Gesendet: Dienstag, 28. November 2017 10:16
An: puuu/ESPiLight
Cc: Dennifly; Author
Betreff: Re: [puuu/ESPiLight] Trying to send Arctech_Switch Signal (#13)

Thanks for using ESPiLight. Is this issue solved and can be closed?

closed 2017-12-05 by puuu