puuu/ESPiLight #17: Disable interrupts while sending

janLo, 2017-11-28:
As sending pulse-trains is a time-critical thing (and as we have only one processor) we should consider to disable interrupts while sending out pulse-trains. This would also prevent that we "see" the echo of our own signal as we send it.

Arduino gives here two possibilities:
* `noInterrupts()`/`interrupts()` to deactivate/activate interrupts globally
* `attachInterrupt()`/`detachInterrupt()` to attach/detach specific ISRs.

I'm not sure what is better. At the moment the ISR that is connected to the receiver pin will get called for every pulse we push to the sender causing the timing to always be slightly off.

puuu, 2017-11-30:
In principle yes, but I did not yet observed issues because of inaccurate pulse length.

Basically, there is code in `ESPiLight::sendPulseTrain(const uint16_t *pulses, int length, int repeats)` to avoid receiving during sending, but this is commented out. Would be good to have this configurable via preprocessor rules.
For some debugging purpose it would good to see your own echo.

I think, `attachInterrupt()`/`detachInterrupt()` will bring no big difference compared to `disableReceiver()`/` enableReceiver()`.

`noInterrupts()`/`interrupts()` would be fine for real time-critical tasks, but we could run into trouble, if the  pulsetrain duration is to long. This options disable wifi interrupts and the watchdog timer can force a reset.

janLo, 2017-11-30:
#21 sould provide a solution.

puuu, 2017-12-05:
solved with #21

closed 2017-12-05 by puuu