puuu/ESPiLight #23: esp32 platform support needs to be tested

puuu, 2017-12-05:
Since v0.12.0, the esp32 platform is supported thanks to @Bacto in #11 . 

Due to lack of esp32 hardware, this is not yet tested. If you can test it, please report your findings in here.

hycday, 2019-01-27:
tried to compile "Receive_Raw" example, got error : 

```
In file included from C:\Users\xxx\Documents\Arduino\libraries\ESPiLight-master\src\ESPiLight.cpp:36:0:

C:\Users\xxx\Documents\Arduino\libraries\ESPiLight-master\src\pilight/libs/pilight/protocols/protocol.h:26:26: fatal error: ../core/json.h: No such file or directory

compilation terminated.

Using library ESPiLight-master at version 0.15.0 in folder: C:\Users\xxx\Documents\Arduino\libraries\ESPiLight-master 
exit status 1
Error compiling for board ESP32 Dev Module.
```

puuu, 2019-02-15:
Hi @hycday , thank you for testing ESPiLight!

Probably, this is not an ESP32 issue. I expect you use the master branch and did not followed the README (same as #12, #25 and #31). Please try with release branch, download from [releases](https://github.com/puuu/ESPiLight/releases) or use the Arduino Library manager.

hycday, 2019-02-16:
you are correct...installing from the Library Manager allowed me to compile successfully :) 

so i tested, and was able to successfully receive a signal that has 2 sync bits, (but with some "issues") : 

the signal from your library is 
```
RAW signal: 250 2583 253 1310 246 316 236 1323 239 318 237 1322 235 320 238 308 238 1326 242 1322 238 313 239 321 231 1323 241 1324 237 326 232 306 239 1329 237 310 239 1328 237 1321 241 314 242 1320 238 316 241 1325 235 318 237 1323 237 320 233 312 239 1329 237 305 243 1323 244 1326 234 311 241 313 238 1326 238 309 240 1327 239 308 239 1328 239 310 238 1328 237 308 240 1324 242 305 243 1327 239 313 236 1321 244 308 240 1330 236 312 236 1327 237 308 242 1327 238 306 243 1320 251 298 251 1318 246 302 247 1326 240 295 252 1318 249 1330 230 311 241 302 248 1316 249 251 257 4451 201 2328 317 5135 
```

while I tested, and actually managed to make it work just an hour ago, to send this signal (and switch the light off successfully): 
```
10765,249,2578,252,1303,249,303,250,1314,236,308,247,1307,246,303,249,305,241,1307,251,1304,249,301,251,297,248,1323,237,1310,242,309,243,298,248,1317,241,300,246,1310,246,1306,249,309,244,1309,244,306,247,1306,246,306,247,1307,245,309,244,303,242,1307,251,301,245,1309,251,1306,246,306,242,305,244,1313,245,300,246,1308,250,311,235,1314,244,301,246,1315,242,324,222,1316,242,306,240,1314,244,300,246,1311,247,313,230,1314,248,310,235,1321,237,300,245,1311,247,307,239,1322,236,305,242,1318,238,304,243,1309,249,312,234,1320,238,1313,238,307,247,304,242,1320,236
```

so your code doesnt catch the first "10765", and then has some extra data bits at the end (you can compare them here https://test.sui.li/oszi/)

i tried with a single data sync bit, and with your code i get : 
```
960 590 449 1107 446 1106 448 1100 453 1104 450 1108 965 590 448 1101 452 1101 971 589 450 1111 961 589 449 1103 969 593 964 587 970 588 969 596 442 1101 457 1095 977 579 458 1101 972 582 457 1096 457 1108 3038 7328 
```

with another library I get 
Decimal: 8542100 (24Bit) Binary: 100000100101011110010100

so looks good :)