puuu/ESPiLight #25: json.h file missing, does not compile

tubalainen, 2017-12-09:
Appointed by the protocol.h file -> #include "../core/json.h" 

json.h does not exist, cannot compile json features using this lib.

`Arduino: 1.8.4 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

C:\Users\torbj\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\protocol.c:27:29: fatal error: protocol_header.h: No such file or directory

 #include "protocol_header.h"

                             ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

puuu, 2017-12-09:
Probably #12 ? Where/how did you got the library? Please install from library manager.

tubalainen, 2017-12-09:
Indeed, my bad! Thanks, compiles like a charm now!

puuu, 2017-12-09:
Thanks, for your comment and much fun with ESPiLight.

closed 2017-12-09 by puuu