puuu/ESPiLight #26: Library does not work with Uno, Nano, Mega

chmasi, 2017-12-17:
Hi,
I am not able to use this library with an Uno, Nano or Mega. 
Is this expected?

janLo, 2017-12-17:
Yes. The library is meant to be used on an ESP8266 or an ESP32.

chmasi, 2017-12-17:
That's a pity, that would also be a great library for the standard arduinos, is there a plan to make it available for the normal Arduinos?

janLo, 2017-12-17:
I doubt that it will fit into atmega controllers. It uses on an esp about 14k of heap space. The atmega 328 has 2k ram at all. Also the atmega devices run only on 16mhz while the esp range from 80 to 250 MHz. So even if it fits in the memory I'm not quite sure if it's fast enough.

puuu, 2017-12-18:
@janLo is correct. ESPiLight use the pilight protocols, a good open source implementation of many 433MHz protocols. But unfortunately, it was not written with micro controller like the AVRs in mind. You are better to go with a more lightweight implementation (see [other similar projects](https://github.com/puuu/ESPiLight#other-similar-projects))  in the README.

closed 2018-01-19 by puuu