puuu/ESPiLight #27: Oregon Temp sensor

tvch, 2018-01-03:
Hello 

Im quite out of my leage when it comes to this stuff, but i've got some sonsors that should work with this newer protocol version. How can i include this branch of pilight in ESPLight? 

https://wiki.pilight.org/oregon

puuu, 2018-01-11:
See also pilight/pilight#270, since this protocol do not send a footer that can be easily detected it required rewriting of core components. Best is to wait for integration into pilight. Unfortunately the development of this protocol seems to stopped.

tvch, 2018-01-11:
Thanks for your answer. I was able compile the corresponding oregon_21.c
file into the piilight source (i disabled the missing log_level function)
but my F007TH are not even supported there.

I think i have to get other sensor, because nobody is in active development
of this...

mlesn, 2018-03-12:
Hello,

Did you manage to setup support for another Oregon sensor?
I have 2x THGR122NX sensors and I wonder if this dev-version of the ordegon decoder will do the job (after compiling it into pilight).

Greetings,
M

> Thanks for your answer. I was able compile the corresponding oregon_21.c
> file into the piilight source (i disabled the missing log_level function)
> but my F007TH are not even supported there.
> 
> I think i have to get other sensor, because nobody is in active development
> of this...

puuu, 2019-03-12:
pilight/pilight#270 is close and the fork is not in active development.

Meanwhile [TFA 2017](https://manual.pilight.org/protocols/433.92/weather/tfa2017.html) has been implemented for pilight. The description refers to a [pdf](http://www.osengr.org/WxShield/Downloads/Weather-Sensor-RF-Protocols.pdf) stating that it is the "Oregon Scientific Protocol versions 1.0, 2.1, 3.0".

ESPiLight v0.16.0 contains this protocol. Maybe it works with your sensors?

I can not test it because I do not have an oregon device. I also do not know if the tfa_2017 protocol can be received with ESPiLight. It is possible that some small changes are required.