puuu/ESPiLight #38: Weather Sensordata never gets published

gsg2820, 2019-02-13:
i have 3 weather Sensors. each sending there data in 1 minute interval.
the status check on received data depends on a second receiving and a string compare. Only then it will be published.
the delay for repeat is to small for reconizing data as a second message. 
Furthermore, the other sensors will send there data and they obviously differ from the first one.

did i miss something? how can i get all three sensor data published without dumping the VALID check

puuu, 2019-02-15:
What kind of weather sensor you are using? Is the protocol supported by pilight? What is the output of the [Receive example](https://github.com/puuu/ESPiLight/blob/master/examples/Receive/Receive.ino)?

Of all protocol I know, the message send by the sender is a repetition (3 or more) of the same pulse train. This pulse trains of the same message are compared for validation. Your 1 minute interval are different messages (of hopefully repeated pulse trains).

gsg2820, 2019-02-15:
Oh, I see.
At least that was my first thought. Though the pulse train is parted on mingaplen. ESPiLight::mingaplen = 5100. Unfortunately the mingaplen isn’t used from each individual protocol.
The actual "SYNC" Pulse in this case is about 4.0ms. And then, as the pulse train was never bin devided I made a wrong turn.
In this case it might be a good idea to write a wrapper protocol to do so.
Thanks for your Answer.

It is a new protocol.
https://forum.pilight.org/showthread.php?tid=2621&page=2