puuu/ESPiLight #39: invalid pulse train message

codegrau, 2019-03-10:
Hi there, I'm using the new integration of ESPiLight in OpenMQTTGateway and getting a lot of "invalid pulse train message" but seems as the rf signal is sent by the gateway. Any idea what's happening here?

https://community.openmqttgateway.com/t/invalid-pulse-train-message/376/2

puuu, 2019-03-12:
@codegrau and @1technophile "invalid pulse train message" is produced by [OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway/blob/307c5e7aa97a03b0d82d198646c221fb62d8bb04/ZgatewayPilight.ino#L111) itself.

I can [track it back](https://github.com/1technophile/OpenMQTTGateway/blob/307c5e7aa97a03b0d82d198646c221fb62d8bb04/ZgatewayPilight.ino#L80) to a call of [`stringToPulseTrain()`](https://github.com/puuu/ESPiLight/blob/master/src/ESPiLight.cpp#L446) which is a ESPiLight function.

[`stringToPulseTrain()`](https://github.com/puuu/ESPiLight/blob/master/src/ESPiLight.cpp#L446) return more detailed error codes that can be used to improve user feedback.

Obviously, the PulseTrain string is invalid. Without providing the PulseTrain string I can not help.

I would recommend to debug the [`message`](https://github.com/1technophile/OpenMQTTGateway/blob/307c5e7aa97a03b0d82d198646c221fb62d8bb04/ZgatewayPilight.ino#L70) variable.

1technophile, 2019-03-12:
Hello @puuu,

Here is the pulse train example we are using:
https://github.com/1technophile/OpenMQTTGateway/wiki/User-guide-PiLight-09#send-data-by-mqtt-to-convert-it-on-rf-signal
Do you have some recommandations about it.

I will add a call to stringToPulseTrain() so as to give better feedback.

puuu, 2019-03-13:
Hello @1technophile
please have a look to puuu/MQTT433gateway@767482efd3715fea8c2deeb913a8b3d5864d9d04 as an example how to provide error messages from `stringToPulseTrain()` return codes.

The example messages you showed, are json stings as used in `rf.send()`, but your issue is in the call of `rf.stringToPulseTrain()`. `rf.stringToPulseTrain()` generates pulse trains from pulse-train strings (e.g. `c:102020202020202020220202020020202200202200202020202020220020202203;p:279,2511,1395,9486@`) and is used as a kind of raw signal support. From [your code](https://github.com/1technophile/OpenMQTTGateway/blob/307c5e7aa97a03b0d82d198646c221fb62d8bb04/ZgatewayPilight.ino#L72), you switch from `rf.send()` to `rf.stringToPulseTrain()` with the condition `!(message && protocol)`. I don't know, when message or protocol is NULL. May you call `rf.stringToPulseTrain()` with an empty message? I would recommend to switch to raw mode only if protocol has a defined value (e.g. `raw`).

1technophile, 2019-03-19:
Hello,

Thanks for these valuable infos, indeed the choice between protocol+message versus raw mode was not well implemented. I took your example for raw error code handling.
https://github.com/1technophile/OpenMQTTGateway/pull/374

@codegrau this issue can be closed I think we can continue discussion on the PR if needed

closed 2019-03-19 by codegrau