puuu/ESPiLight #46: wrong protokoll selection?

ansim58, 2019-12-30:
Hi, i want to read signals from a rainmeter: 
- It's a TFA 30.3161 (https://www.amazon.de/TFA-Dostmann-30-3161-Regensensor-FS-NEUTR/dp/B00FYSUATO/ref=pd_rhf_se_p_img_4?_encoding=UTF8&psc=1&refRID=J7QA97F1ZFG5QHR2JS4Y).
- It sends temperature and rain-data to a weatherstation.
- I receive/send signals on 433Mhz with this couple: (https://www.az-delivery.de/products/433-mhz-modul).
- I am testing on a ESP8266 (D1 mini) using the 'Basic ESPiLight pilight_debug example'.
Hope I got all the features listed...
The message received every minute is i.e.: 14:15:05.397 -> RF signal arrived [alecto_wsd17][4095] (1) Message1:{"id":4095,"temperature":409.5}
14:15:05.397 -> --[RESULTS]--
14:15:05.397 -> 
14:15:05.397 -> time:		28988 ms
14:15:05.397 -> hardware:	ESPiLight
14:15:05.397 -> pulse:		15
14:15:05.397 -> rawlen:		74
14:15:05.397 -> pulselen:	267
14:15:05.397 -> 
14:15:05.397 -> Raw code:
14:15:05.397 -> 449 4066 449 4065 453 4069 448 1989 451 1997 442 1992 449 1994 445 1999 441 1999 441 2000 440 2002 439 2002 438 2002 440 2004 436 2007 433 2003 437 2009 432 4080 437 4082 436 4082 436 2004 436 2005 435 2006 434 2007 434 4082 436 4078 439 2005 436 4084 433 2002 442 2001 436 2008 432 2015 425 2006 435 2012 429 2012 428 2137 426 9090 
As you can see, there is something wrong, at least with the temperature: There are definately not 409.5 Deg in my room :-)
The other point is the ID: 4095 is quite similar to the temp: 409.5
That is why I think,  the sketch ist propably using the wrong protocoll: alecto_wsd17
It might be better to use a TFA-Protokol as provided with the pilight-library.
Now the question: Is it possible to determine the protokill to be used from ESPiLight?
Thanks in advance, best regards, Andreas

ansim58, 2020-01-12:
Hi!
Nobody has any idea???
Is it possible to determine the protokill to be used from ESPiLight?
Thank you all! 
Andreas

puuu, 2020-01-15:
ESPiLight uses all activated (default: all available) protocols to parse the rf messages. If the rf message can be analyzed in some way (e.g. pulse lengths or number of pulses, see `validate()` of the respective protocol implementation) using one or more protocols, then the callback function is called with these results (if several protocols match, then the callback function is called multiple times). The user is asked to recognize which is a meaningful protocol.

In your case, only the "alecto_wsd17" protocol parsed the RF message, but as you noted, it is not the correct one.

Your rf message has a raw length of 74 pulses, but tfa requires [at least 76 pulses](https://github.com/pilight/pilight/blob/master/libs/pilight/protocols/433.92/tfa.c).

For [new protocols](https://github.com/puuu/ESPiLight#new-protocols) please report directly to pilight.

closed 2020-01-15 by puuu