puuu/ESPiLight #51: Nexus protocol

piano-thomas, 2020-09-29:
Hi all,

I am trying to implement the Nexus protocol into ESPilight, but cannot get it to work. I tried several things, but I am probably not overseeing everything. Apparently, this protocol has only just been implemented in Pilight if I am correct. I checked several forums, but cannot get it to work. In the end I am trying to get an Auriol weather sensor to send data to an ESP8266 device.

Hope someone can help!

Thanks!

enwi, 2020-10-21:
I am also trying to get it work so I pipe the data through `OpenMQTTGateway` to `HA`. The code compiles fine when adding `nexus.c` and `nexus.h`, but no nexus data is shown in the console.

piano-thomas, 2020-10-21:
> I am also trying to get it work so I can pipe the data through `OpenMQTTGateway` to `HA`. The code compiles fine when adding `nexus.c` and `nexus.h`, but no nexus data is shown in the console.

Yes, I did exactly the same, but that didn't help me getting data in the serial monitor. I hope someone can help us!

enwi, 2020-10-21:
I am getting somewhere:
```
pulses: 79 possible protocol: nexus
pulses: 81 possible protocol: nexus
pulses: 84 possible protocol: nexus
pulses: 77 possible protocol: nexus
```

piano-thomas, 2020-10-21:
@enwi did you change anything in the Receive.ino file?

enwi, 2020-10-21:
Nope, I think that Arduino did not compile the changed files and somehow I got it to compile it, but now I need to figure out why it is not parsing correctly

enwi, 2020-10-21:
Debugging the parsing of the nexus signal it seems that the ESP resolution of captured pulses is too low

enwi, 2020-10-21:
`validate()` will find a valid pulse, but then during parsing the second pulse is almost always wrong

enwi, 2020-10-21:
Forget what I said, I got it working
```
RF signal arrived [nexus][200] (0) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (2) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
Valid message: [nexus] {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (3) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (3) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (3) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (3) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (3) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (3) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (3) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (3) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [nexus][200] (0) {"id":200,"channel":0,"temperature":24.5,"humidity":56,"battery":0}
RF signal arrived [clarus_switch][E11] (0) {"id":"E11","unit":45,"state":"off"}
```

piano-thomas, 2020-10-21:
You are amazing! No clue what you have exactly done, but it works!!

19:58:28.175 -> RF signal arrived [nexus][81] (0) {"id":81,"channel":2,"temperature":23.4,"humidity":61,"battery":1}
19:58:28.246 -> RF signal arrived [nexus][81] (2) {"id":81,"channel":2,"temperature":23.4,"humidity":61,"battery":1}
19:58:28.246 -> Valid message: [nexus] {"id":81,"channel":2,"temperature":23.4,"humidity":61,"battery":1}
19:58:28.315 -> RF signal arrived [nexus][81] (3) {"id":81,"channel":2,"temperature":23.4,"humidity":61,"battery":1}
19:58:28.382 -> RF signal arrived [nexus][81] (3) {"id":81,"channel":2,"temperature":23.4,"humidity":61,"battery":1}
19:58:28.483 -> RF signal arrived [nexus][81] (3) {"id":81,"channel":2,"temperature":23.4,"humidity":61,"battery":1}
19:58:28.553 -> RF signal arrived [nexus][81] (3) {"id":81,"channel":2,"temperature":23.4,"humidity":61,"battery":1}
19:58:28.623 -> RF signal arrived [nexus][81] (3) {"id":81,"channel":2,"temperature":23.4,"humidity":61,"battery":1}

enwi, 2020-10-21:
Nice, the issue was that the signals were dropped due to a default defined length, which were needed for nexus to work. So i changed the code that it will check if it needs to adjust those values in order to make sure all needed signal lengths are catched.

piano-thomas, 2020-10-21:
> Nice, the issue was that the signals were dropped due to a default defined length, which were needed for nexus to work. So i changed the code that it will check if it needs to adjust those values in order to make sure all needed signal lengths are catched.

Great work, thanks a lot!

closed 2020-10-21 by piano-thomas

marekorok, 2020-10-25:
Hello,
I've been trying like you with the nexus protocol with my 3 devices, from which 2 all exactly the same (DG-R8H, the other different unbranded) and found its parameters are not very reliable, even within the same devices.

I started playing with the PULSE_TOLERANCE and with value 100 I got one device working, when increased to 150 I got two of them working (one unbranded, one DG-R8H), when I put 200 I got all of them working.

puuu, 2020-10-27:
ESPiLight [v0.17.0](/puuu/ESPiLight/releases) is now released and should have working nexus support. The version will soon be available in the Arduino library manager.

For problems with protocols please report to pilight directly. Please also see the [New protocols sectrion in the README](https://github.com/puuu/ESPiLight#new-protocols).