puuu/ESPiLight #60: alecto_ws1700 protocol not received with version 0.17.0

LeoVe, 2021-05-30:
Hi,

Thanks for the great library !

I have used the library on wemos D1 mini pro to receive temperature and humidity from a Konig Weatherstation using the alecto_ws1700 protocol.

It works fine with version 0.16.2, but the messages for this protocol are not received with version 0.17.0 of the library.

I loaded the libraries in Arduino v1.8.13 with standard library manager.

Thanks!

dkluivingh, 2021-09-23:
Hi,

I got the same problem with alecto_wx500 protocol, works fine with version 0.16.2, but not on version 0.17.0.
Running on OpenMQTTGateway version 0.9.8 with an esp32 and a RXB6 receiver.

If you need more information and I can help with anything, I am available.

Thanks.

LeoVe, 2021-09-23:
You're right, thanks dkluivingh

I deinstalled version 0.17.0 and installed  version 0.16.2 and then it worked fine yes.

Just mentioned it as a good to know for users and developers.

Maybe a future version might include the missing protocols again?

Kind regards.

dkluivingh, 2021-09-23:
Hi LeoVe,

Yeah did the same here.

Greetz