puuu/ESPiLight #62: Error compiling for ESP8266

tichachm, 2021-07-20:
Hi,

I try to use this library with ESP 8266 Board under Arduino IDE.

ESPiLight: 1.7.0
ESP8266 (NodeMCU1.0): 3.0.1

In addition to my  own I tested the example receive sketch, same problem here.

Keeps getting this error since I switched to ESP lib version 3.0.0:
`c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\rev_v3.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: multiple definition of `dso'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\alecto_ws1700.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\rsl366.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/rsl366.h:24: multiple definition of `rsl366'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\protocol.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/rsl366.h:24: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\rsl366.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: multiple definition of `dso'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\alecto_ws1700.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\sc2262.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/sc2262.h:24: multiple definition of `sc2262'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\protocol.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/sc2262.h:24: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\sc2262.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: multiple definition of `dso'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\alecto_ws1700.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\secudo_smoke.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/secudo_smoke.h:24: multiple definition of `secudo_smoke'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\protocol.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/secudo_smoke.h:24: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\secudo_smoke.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: multiple definition of `dso'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\alecto_ws1700.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\selectremote.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/selectremote.h:24: multiple definition of `selectremote'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\protocol.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/selectremote.h:24: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\selectremote.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: multiple definition of `dso'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\alecto_ws1700.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\silvercrest.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/silvercrest.h:24: multiple definition of `silvercrest'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\protocol.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/silvercrest.h:24: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\silvercrest.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: multiple definition of `dso'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\alecto_ws1700.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: first defined here
c:/users/dummy/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/3.0.3-gcc10.3-9bcba0b/bin/../lib/gcc/xtensa-lx106-elf/10.3.0/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\smartwares_switch.c.o:C:\Users\dummy\Documents\Arduino\libraries\ESPiLight\src\pilight\libs\pilight\protocols\433.92/smartwares_switch.h:14: multiple definition of `smartwares_switch'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\protocol.c.o:C:\Users\dummy\Do [... cut ...] ld_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\x10.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: multiple definition of `dso'; C:\Users\dummy\AppData\Local\Temp\arduino_build_341315\libraries\ESPiLight\pilight\libs\pilight\protocols\433.92\alecto_ws1700.c.o:c:\users\dummy\documents\arduino\libraries\espilight\src\pilight\libs\pilight\core/dso.h:36: first defined here
collect2.exe: error: ld returned 1 exit status`

Thank you

h2zero, 2021-08-07:
Any luck resolving this? We have the same problem when updating the 8266 framework to 3.1.0 in platformio.

tichachm, 2021-08-08:
No, I am just using ESP lib < 3.0.0 to compile my pilight sketch

EricG78, 2022-01-11:
In Arduino IDE, with ESP8266 3.0.2 environment, the link errors disappeared after:

1. Commenting the line `#include "protocol_header.h"` in the file protocols.c
2. Commenting the line `#include "../../core/dso.h"` in all the c files in 433.92 directory (for instance with the command `sed -i -e '/dso\.h/s/^/\/\//' *.c`

davidfurey, 2022-11-25:
I think this is an issue with piilight rather than ESPiLight.  I've opened https://github.com/pilight/pilight/pull/469 which should fix the upstream issue

yxmnas, 2023-05-05:
With ESP8266 3.0.2 I used the modified version of pilight (https://github.com/pilight/pilight/pull/469/commits/20800cc7e067155b219a85d7bcd17446a316e452) and commented the dso-line as suggested by EricG78 with Notepad++.